mr-manuel/venus-os_dbus-mqtt-battery #1: Request for all cells voltages displaying feature.

xury77, 2023-03-10:
Hi mr-manuel.
First of all I'd like to send you very big thanks for this feature. This is what I looking for very long time!
I'd like to asking you about new feature such as displaying all cell voltages in details tab. 
I've got my custom  battery with  custom BMS and I can send all cell voltages to MQTT broker. It will be very nice to see these all voltages in Victron. It would also be useful to display delta voltage value.
I can also confirm that it works correctly on CerboGX.

mr-manuel, 2023-03-10:
Hi @xury77,
how many cell voltages do you have?

xury77, 2023-03-10:
It would be great. I've used dbus-serialbattery in the past. Plugin is ok, but Daly BMS not so. 
Yet one question: How to add second battery instance?

mr-manuel, 2023-03-10:
Just released a complete new version.

Check the [Multiple instances](https://github.com/mr-manuel/venus-os_dbus-mqtt-battery/tree/v1.0.0#multiple-instances) section in the README.

Please give me a feedback, if all is working on your side.

closed 2023-03-10 by mr-manuel

xury77, 2023-03-13:
Awesome. Working like a harm.

mr-manuel, 2023-03-13:
@xury77 harm doesn't mean anything good.

xury77, 2023-03-14:
Hi  **@mr-manuel** its me again -  your annoying issue maker :)
Last question (I hope): Missing HighCellVoltage in Alarms is not provided by Victron ?
I see "LowCellVoltage" but no "HighCellVoltage"

mr-manuel, 2023-03-14:
Hi @xury77,
indeed this alarm is not provided by Victron. See the [dbus documentation](https://github.com/victronenergy/venus/wiki/dbus#battery). Also if you check the QML files you cannot find this value.

xury77, 2023-03-14:
Yes you are right. It seems they don't control this value in their BMS. Or this situation may never occur.

xury77, 2023-03-14:
Sorry one  more question about "MinVoltageCellId" and others. You wrote:

> "MinVoltageCellId": "C3",             --> String - ID of the cell with the lowest voltage - if empty, than gets calculated when elements in "Voltages" are present
        "MinCellVoltage": 3.392,              --> Volt - Of the cell with the lowest voltage - if empty, than gets calculated when elements in "Voltages" are present

> 
What exactly mean "if empty"? I asking because I tried:
Not to send this property
Sending as null

> msg.payload.System.MinVoltageCellId = null;
msg.payload.System.MinCellVoltage = null;
msg.payload.System.MaxVoltageCellId = null;
msg.payload.System.MaxCellVoltage = null;``

Sending string properties as empty string:

> msg.payload.System.MinVoltageCellId = ""; 
   msg.payload.System.MinCellVoltage = null;

And still no values on Detalis page and also no Voltage sum nor Min/Max/Diff
But cell voltages are present.
Perhaps I have to send also empty messages to MinTemperatureCellId and others witch I not use?

mr-manuel, 2023-03-14:
I changed the desciption to "if missing in the JSON". Can you post the JSON that you sent?

xury77, 2023-03-14:
My payload:
``{"Dc":{"Power":0,"Voltage":52.61,"Current":0,"Temperature":19},"InstalledCapacity":400,"ConsumedAmphours":-340.2,"Capacity":124.5,"Soc":19,"Alarms":{"LowVoltage":0,"HighVoltage":0,"LowSoc":0,"LowTemperature":0,"HighTemperature":0},"Info":{"ChargeRequest":false,"MaxChargeVoltage":58.9,"MaxChargeCurrent":400,"MaxDischargeCurrent":800},"History":{"ChargeCycles":7,"TotalAhDrawn":1313.7},"System":{"NrOfModulesOnline":1,"NrOfModulesOffline":0},"Voltages":{"Cell1":3.281,"Cell2":3.29,"Cell3":3.286,"Cell4":3.289,"Cell5":3.298,"Cell6":3.276,"Cell7":3.293,"Cell8":3.288,"Cell9":3.296,"Cell10":3.295,"Cell11":3.298,"Cell12":3.294,"Cell13":3.292,"Cell14":3.296,"Cell15":3.296,"Cell16":3.297},"Io":{"AllowToCharge":1,"AllowToDischarge":1}}``

mr-manuel, 2023-03-14:
> BTW "Alarms" can't be a bools as it a three state values: 0 -OK 1-Warning 2-Alarm

Nice catch, thanks. I corrected it.

`"ConsumedAmphours": -340.2,` should not be negative, but this is not causing the problem.

The problem was a division by zero when current = 0 in the Time to Go calculation. I publish a new release.

xury77, 2023-03-14:
I can confirm - working :)
btw. I get this negative value directly from SmartShunt so I use Math.abs() to convert it.
Once again Very Thanks for this handy plugin.

pkkrusty, 2026-09-16:
> Hi [@xury77](https://github.com/xury77), displaying the cell voltages is not a native feature of Venus OS, but I will add this soon the same way it is the [dbus-serialbattery](https://github.com/Louisvdw/dbus-serialbattery) driver.

Did this ever happen? I'm using the full MQTT spec, and json message includes all cell voltages, but I don't see cell voltages in detail menu, just min and max.