mr-manuel/venus-os_dbus-mqtt-battery #2: Example in config.ini is lowercase, warning seems to be Case-sensitiv

pos-ei-don, 2023-03-15:
The example in the config.ini is lowercase
> { "dc": { "power": 321.6, "voltage": 52.6  }, "soc": 63 }

The resulting warning is mixed-Case 
> {"Dc":{"Power":321.6,"Voltage":52.7},"Soc":63}

The first one throws a warning (:Received JSON doesn't contain minimum required values), the second shows this in the logs
> *** starting dbus-mqtt-battery ***

What am I doing wrong? Did I unserstand correctly that this driver needs one single topic with this JSON-Value?

I also tried seperate topics with this hirarchy, but did not work either.

Thank you a lot for this driver!

mr-manuel, 2023-03-15:
> The example in the config.ini is lowercase
> 
> > { "dc": { "power": 321.6, "voltage": 52.6  }, "soc": 63 }
> 
> The resulting warning is mixed-Case
> 
> > {"Dc":{"Power":321.6,"Voltage":52.7},"Soc":63}

Thanks, I corrected it.

> Sending the first JSON to the topic throws a warning (:Received JSON doesn't contain minimum required values), the second shows this in the logs
> 
> > *** starting dbus-mqtt-battery ***

If you get no other messages, all is ok and the driver is running fine. To be sure, that all is OK you can temporarily increase the log level to `logging = INFO` in your `config.ini`. Don't forget to reset it after to `WARNING`, else you spam your logfile.

> What am I doing wrong? Did I unserstand correctly that this driver needs one single topic with this JSON-Value?

Yes, one topic with the complete JSON values.

pos-ei-don, 2023-03-16:
Perfekt, DANKE!
Jetzt bekomme ich als Fehlermeldung:
"/dc/voltage" with value "53.8" is not valid

Macht hier der Punkt das Problem oder sollte der Wert im JSON mit " als Text übergeben werden? Ich werde hier noch etwa sversuchen.
Auch hier der Hinweis, dass diese Fehlermeldung in lowercase ausgegeben wird und somit falsch verstanden werden kann ;-)

mr-manuel, 2023-03-16:
Es muss auch `Dc` und `Voltage` sein. Alles klein mag er nicht. Kannst du deinen Flow und das generierte JSON posten?

closed 2023-03-16 by pos-ei-don