mr-manuel/venus-os_dbus-mqtt-battery #3: Victron MPPT does not get "BMS controlled, extend "minimal required JSON payload"

pos-ei-don, 2023-03-16:
The "minimal required JSON payload"
is not enough to set an MPPT in "network/BMS controlled mode".

I suggest to add a line in the config for 
 "minimal required JSON payload to allow BMS controlled MPPT".

I currently did not figure out what I needed to add, but I will try to test and find out.
If someone or you (@mr-manuel ) know what is additionally necessary, I would help me a lot.

mr-manuel, 2023-03-16:
Hello,

unfortunately I have no Victron MPPT nor do I know what values are needed. If you find out, I will add it to the README.

I can only guess and maybe one or all of this values are needed, but I can also be completely wrong.

```json
"Info": {
    "MaxChargeVoltage": 55.2,
    "MaxChargeCurrent": 80,
    "MaxDischargeCurrent" : 80
}
```

If you provide the full JSON it works?

pos-ei-don, 2023-03-16:
Yes, thats it!
I didn't test if a single value is enough, i will test this later.
When I set all 3 values, "BMS controlled" immediately switches to "on".

Great, thank you a lot!

closed 2023-03-16 by pos-ei-don

mr-manuel, 2023-03-16:
Probably only this two are needed, since the MPPT do only load the battery.

```json
"Info": {
    "MaxChargeVoltage": 55.2,
    "MaxChargeCurrent": 80
}
```

mr-manuel, 2023-03-16:
@pos-ei-don can you try with only this two and tell me the result?