mr-manuel/venus-os_dbus-mqtt-battery #4: Ability to set battery offline/online

xury77, 2023-03-27:
Hi again :)
I thought it would be nice to be able to set the battery to offline mode when there are no messages to the mqtt broker or certain battery topic.
Online/offline management could be dealt with separately, e.g. via Nodered flow - no new messages for 1 minute - send offline flag.
I tested NrOfModulesOnline but it doesn't seem to work like that. On Devices page still is connected.

mr-manuel, 2023-03-27:
Hi, what is the goal you try to achieve?
1. You only want to know, if the battery is online or not if checking in Venus OS
2. You want that the battery disconnects (same as if you pull the plug), if no message after x seconds is received
3. You want to get a warning when the battery is offline

mr-manuel, 2023-03-27:
> Yesterday the mqtt battery stopped updating its values (dont know why) and my chargers kept charging with 300A, while the battery Pack already had soc>98%

Did you check the logs? You can find in the README for how to do that.

pos-ei-don, 2023-03-27:
My goal is to Set the Inverters/Chargers into a safe state when the connection gets lost.
If inunregister the battery, the Multiplus stops charging. This seems to be a good solution, so i`d suggest disconnect Ing the mqtt battery if the connection to the mqtt Broker is disconnect >10seconds or if i dont get any values for over 60s.
The 60s watchdog time is also implemented in VenusOS for safety important values (like external Controller setpoint).

pos-ei-don, 2023-03-27:
> > Yesterday the mqtt battery stopped updating its values (dont know why) and my chargers kept charging with 300A, while the battery Pack already had soc>98%
> 
> Did you check the logs? You can find in the README for how to do that.

No, was not at home and had no chance.
The mqtt Broker restarted, but i dont know if this has any impact on connected mqtt Devices.

mr-manuel, 2023-03-27:
You can also check the logs now. You should find something even if it was yesterday.

mr-manuel, 2023-03-28:
I just released a new version.

You can now specify a timeout in the `config.ini` after which the driver exits, if no new MQTT message is received. If you installed the driver, then it gets automatically restarted and connects again when a new MQTT message is received.

pos-ei-don, 2023-03-28:
> You can also check the logs now. You should find something even if it was yesterday.

Unfortunately, by updating to 1.0.2, i overwrote the logfile. sorry.
I will check the logs if this ever happens again.
What I remember is that the mqtt was restarted, but filled with newer values.
The driver did not get the new values. But this already should be solved in 1.0.2,
so i have to say THANK YOU A LOT!!!

mr-manuel, 2023-03-28:
With `cat /data/log/dbus-mqtt-battery/* | tai64nlocal` you see all logfiles, also the history.

pos-ei-don, 2023-03-28:
no, this file is empty. I think i deleted it during the update. 
is my mistake!

xury77, 2023-03-28:
I'm just testing. It seems to be working fine.