mr-manuel/venus-os_dbus-mqtt-battery #6: Auto-Restart did not work

pos-ei-don, 2023-03-29:
Here is the Logs i found.

If I understand right, the driver did not recive any MQTT Updates.
He tried to auto-restart at 09:14 which failed.
My manual restart at 09:53 succeded.

```
2023-03-29 09:10:32.975302500 ERROR:root:Driver stopped. Timeout of 60 seconds exceeded, since no new MQTT message was received in this time.
2023-03-29 09:10:33.085773500 *** starting dbus-mqtt-battery ***
2023-03-29 09:14:04.473393500 Traceback (most recent call last):
2023-03-29 09:14:04.473397500   File "/data/etc/dbus-mqtt-battery/dbus-mqtt-battery.py", line 381, in _update
2023-03-29 09:14:04.473401500     self._dbusservice[setting] = data['value']
2023-03-29 09:14:04.473404500   File "/data/etc/dbus-mqtt-battery/ext/velib_python/vedbus.py", line 157, in __setitem__
2023-03-29 09:14:04.473408500     self._dbusobjects[path].local_set_value(newvalue)
2023-03-29 09:14:04.473411500   File "/data/etc/dbus-mqtt-battery/ext/velib_python/vedbus.py", line 499, in local_set_value
2023-03-29 09:14:04.473488500     changes = self._local_set_value(newvalue)
2023-03-29 09:14:04.473492500   File "/data/etc/dbus-mqtt-battery/ext/velib_python/vedbus.py", line 510, in _local_set_value
2023-03-29 09:14:04.473496500     'Text': self.GetText()
2023-03-29 09:14:04.473498500   File "/data/etc/dbus-mqtt-battery/ext/velib_python/vedbus.py", line 589, in GetText
2023-03-29 09:14:04.473502500     return self._gettextcallback(self.__dbus_object_path__, self._value)
2023-03-29 09:14:04.473530500   File "/data/etc/dbus-mqtt-battery/dbus-mqtt-battery.py", line 64, in <lambda>
2023-03-29 09:14:04.473534500     _a = lambda p,  v: (str("%.2f" % v) + 'A')
2023-03-29 09:14:04.473537500 TypeError: must be real number, not dict
2023-03-29 09:53:46.758113500 *** starting dbus-mqtt-battery ***

```

mr-manuel, 2023-03-29:
Can you post the JSON you sent when the driver crashed? It looks like you send an object instead of a number as amps or is this calculated by the driver?

pos-ei-don, 2023-03-29:
I do not set any amps, they are calculated by the driver!

Here is the payload I send.

Offtopic for this thread but maybe interesting for you:
If I don't set "MaxDischargeCurrent" when the MPPTs are BMS-Connected, I get an Low-Voltage Alarm in
VenusOS after restart. Regardless the actual Voltage. Setting MaxDischargeCurrent=0  or higher is ok!

```
{
  "Dc": {
    "Voltage": 53.8,
    "Power": 435
  },
  "Soc": 90,
  "Info": {
    "MaxChargeVoltage": 56.3,
    "MaxChargeCurrent": 10,
    "MaxDischargeCurrent": 200
  }
}
```

mr-manuel, 2023-03-29:
The auto restart works, when no MQTT message is received, but in the case above it's another issue.

I added some error handling. Could you download the script from the main branch and test it? So the next time it happens we see what caused the issue.

mr-manuel, 2023-03-29:
> Offtopic for this thread but maybe interesting for you:
> If I don't set "MaxDischargeCurrent" when the MPPTs are BMS-Connected, I get an Low-Voltage Alarm in
> VenusOS after restart. Regardless the actual Voltage. Setting MaxDischargeCurrent=0 or higher is ok!

I add all three values to `Minimum required that the MTTP is controlled by BMS` in the README

mr-manuel, 2023-03-29:
Are you defining `MaxChargeCurrent` and `MaxDischargeCurrent` statically or dynamically? It seems that at the moment of the error an object was passed instead of a value.

I commited an other change which includes type checks before assigning the values. If you already downloaded the script, please redownload it from the master branch :)

pos-ei-don, 2023-03-29:
Dynamicaly, but there is no object in the Flow. So this shouldn't be possible, and I have ever seen this when looking at my Flow, but i will take a deeper look

mr-manuel, 2023-03-29:
With the latest script it will output an error and show the value that was sent. So just take the latest script and check the logs after a few hours.

`WARNING:root:Received key "/Info/MaxDischargeCurrent" with value "{'test': 1}" is not valid`

pos-ei-don, 2023-03-29:
Thank you. just upgraded!! Will monitor the Device the next days!

Do you use a cerbo?
When using restart.sh under current venusos 2.93,
I get this error:
> kill: usage: kill [-s sigspec | -n signum | -sigspec] pid | jobspec ... or kill -l [sigspec]

pos-ei-don, 2023-03-29:
The difference Was that the running Script was the Older Version. Fixed it by manually killing it, which worked.

closed 2023-04-04 by mr-manuel