mr-manuel/venus-os_dbus-mqtt-battery #7: ERROR:config.ini file not found. Copy or rename the config.sample.ini to config.ini.

harald7, 2023-04-04:
Hi Manuel
Thanks alot for sharing your software.  I have already view times tried to run your driver but I get alway this error message. Venus OS is fresh installed (2.93) on a RPI 4 and nothing is connected. MQTT is running. The config.ini is in the folder.
[image]

What I'm doing wrong?

Best regards

Harald

mr-manuel, 2023-04-04:
Hi @harald7,

can you post your `config.ini`? Please remove the password before, if set.

Can you also post the result of `ls -l /data/etc/dbus-mqtt-battery`?

harald7, 2023-04-04:
Sure!
[image]
`; CONFIG FILE
; GitHub reporitory: https://github.com/mr-manuel/venus-os_dbus-mqtt-battery
; remove semicolon ; to enable desired setting

[DEFAULT]
; Set logging level
; ERROR = shows errors only
; WARNING = shows ERROR and warnings
; INFO = shows WARNING and running functions
; DEBUG = shows INFO and data/values
; default: WARNING
logging = DEBUG

; Device name
; default: MQTT Battery
device_name = MQTT Battery

; Device VRM instance
; default: 41
device_instance = 41

; Specify after how many seconds the driver should exit (disconnect), if no new MQTT message was received
; default: 60
; value to disable timeout: 0
timeout = 60

[MQTT]
; IP addess or FQDN from MQTT server
broker_address = 192.168.60.2

; Port of the MQTT server
; default plaintext: 1883
; default TLS port: 8883
broker_port = 1883

; Enables TLS
; 0 = Disabled
; 1 = Enabled
;tls_enabled = 1

; Absolute path to the Certificate Authority certificate file that is to be treated as trusted by this client
;tls_path_to_ca = /data/keys/mosquitto.crt

; Disables verification of the server hostname in the server certificate
; 0 = Disabled
; 1 = Enabled
;tls_insecure = 1

; Username used for connection
;username = myuser

; Password used for connection
;password = mypassword

; Topic where the meters data as JSON string is published
; minimum required JSON payload: { "Dc": { "Power": 321.6, "Voltage": 52.6  }, "Soc": 63 }
topic = enphase/battery

{
    "Dc": {
        "Power": 321.6,                       
        "Voltage": 52.7,                      
        "Current": 6.10,                      
        "Temperature": 23                     
    },
    "InstalledCapacity": 200.0,               
    "ConsumedAmphours": 74.5,                 
    "Capacity": 125.5,                        
[... 84 lines cut ...]
        "Cell9":  0,                          
        "Cell10": 0,                          
        "Cell11": 0,                          
        "Cell12": 0,                         
        "Cell13": 0,                          
        "Cell14": 0,                         
        "Cell15": 0,                          
        "Cell16": 0,                          
        "Cell17": 0,                          
        "Cell18": 0,                          
        "Cell19": 0,                          
        "Cell20": 0,                          
        "Cell21": 0,                          
        "Cell22": 0,                          
        "Cell23": 0,                          
        "Cell24": 0                           
    },
    "Io": {
        "AllowToCharge": 0,                   
        "AllowToDischarge": 0,                
        "AllowToBalance": 0,                  
        "ExternalRelay": 0                    
    }
}
`

mr-manuel, 2023-04-04:
> Could you also download the `dbus-mqtt-battery.py` from the master branch and use it? I changed something for better error handling to understand better the problem. You can run this command to do so: `wget -O /data/etc/dbus-mqtt-battery/dbus-mqtt-battery.py https://raw.githubusercontent.com/mr-manuel/venus-os_dbus-mqtt-battery/master/dbus-mqtt-battery/dbus-mqtt-battery.py`

Have you uploaded the script by FTP/SFTP or downloaded directly in Venus OS?

harald7, 2023-04-04:
I have uploaded with WinSCP.

harald7, 2023-04-04:
`2023-04-04 18:34:56.325368500 ERROR: Source contains parsing errors: '/data/etc/dbus-mqtt-battery/config.ini'
2023-04-04 18:34:56.325374500   [line 59]: '{\n'
2023-04-04 18:34:56.325377500   [line 176]: '}\n'
2023-04-04 18:34:56.325380500 If config.ini file is not found then copy or rename the config.sample.ini to config.ini. The driver restarts in 60 seconds.
2023-04-04 18:34:56.407045500 *** starting dbus-mqtt-battery ***
`

mr-manuel, 2023-04-04:
Thats the problem. The line endings where changed during upload. From `\n` to `\r\n`.

See https://winscp.net/eng/docs/faq_line_breaks.

harald7, 2023-04-04:
Should I download direct in VenusOS?
If yes which version?

mr-manuel, 2023-04-04:
Yes, download directly in Venus OS. You can download from the master branch.

harald7, 2023-04-04:
Like this?
`wget -O /data/etc/dbus-mqtt-battery/dbus-mqtt-battery.py https://raw.githubusercontent.com/mr-manuel/venus-os_dbus-mqtt-battery/master/venus-os_dbus-mqtt-battery-master.zip`

mr-manuel, 2023-04-04:
Try this, haven't tested it.

```bash
cd /tmp

wget https://github.com/mr-manuel/venus-os_dbus-mqtt-battery/archive/refs/heads/master.zip
unzip master.zip

cp -rf /tmp/venus-os_dbus-mqtt-battery-master/etc/dbus-mqtt-battery/ /data/etc

chmod +x /data/etc/dbus-mqtt-battery/*.sh
chmod +x /data/etc/dbus-mqtt-battery/*.py
chmod +x /data/etc/dbus-mqtt-battery/service/run
chmod +x /data/etc/dbus-mqtt-battery/service/log/run

bash /data/etc/dbus-mqtt-battery/install.sh
```

harald7, 2023-04-04:
Hi Manuel

I managed to install the driver with your instructions. Thank you for that.
cp -rf /tmp/venus-os_dbus-mqtt-battery-master/**etc**/dbus-mqtt-battery/ /data/etc
The first etc was wrong.

That's what I have done:

cd /tmp

wget https://github.com/mr-manuel/venus-os_dbus-mqtt-battery/archive/refs/heads/master.zip

unzip master.zip

cp -rf /tmp/venus-os_dbus-mqtt-battery-master/dbus-mqtt-battery/ /data/etc

chmod +x /data/etc/dbus-mqtt-battery/*.sh

chmod +x /data/etc/dbus-mqtt-battery/*.py

chmod +x /data/etc/dbus-mqtt-battery/service/run

chmod +x /data/etc/dbus-mqtt-battery/service/log/run

cd /data/etc/dbus-mqtt-battery

ls

mv config.sample.ini config.ini

nano config.ini #Made the adjustments and insert the json

bash /data/etc/dbus-mqtt-battery/install.sh

svstat /service/dbus-mqtt-battery

tail -n 100 -f /data/log/dbus-mqtt-battery/current | tai64nlocal
 
Unfortunately I get the same error.

`2023-04-04 20:37:55.075887500 *** starting dbus-mqtt-battery ***
2023-04-04 20:38:55.900223500 ERROR: Source contains parsing errors: '/data/etc/dbus-mqtt-battery/config.ini'
2023-04-04 20:38:55.900229500   [line 59]: '{\n'
2023-04-04 20:38:55.900232500   [line 176]: '}\n'
2023-04-04 20:38:55.900234500 If config.ini file is not found then copy or rename the config.sample.ini to config.ini. The driver restarts in 60 seconds.
`

mr-manuel, 2023-04-05:
> ```ini
> ; CONFIG FILE
> ; GitHub reporitory: https://github.com/mr-manuel/venus-os_dbus-mqtt-battery
> ; remove semicolon ; to enable desired setting
> 
> [DEFAULT]
> ; Set logging level
> ; ERROR = shows errors only
> ; WARNING = shows ERROR and warnings
> ; INFO = shows WARNING and running functions
> ; DEBUG = shows INFO and data/values
> ; default: WARNING
> logging = DEBUG
> 
> ; Device name
> ; default: MQTT Battery
> device_name = MQTT Battery
> 
> ; Device VRM instance
> ; default: 41
> device_instance = 41
> 
> ; Specify after how many seconds the driver should exit (disconnect), if no new MQTT message was received
> ; default: 60
> ; value to disable timeout: 0
> timeout = 60
> 
> [MQTT]
> ; IP addess or FQDN from MQTT server
> broker_address = 192.168.60.2
> 
> ; Port of the MQTT server
> ; default plaintext: 1883
> ; default TLS port: 8883
> broker_port = 1883
> 
> ; Enables TLS
> ; 0 = Disabled
> ; 1 = Enabled
> ;tls_enabled = 1
> 
> ; Absolute path to the Certificate Authority certificate file that is to be treated as trusted by this client
> ;tls_path_to_ca = /data/keys/mosquitto.crt
> 
> ; Disables verification of the server hostname in the server certificate
> ; 0 = Disabled
> ; 1 = Enabled
> ;tls_insecure = 1
> 
> ; Username used for connection
> ;username = myuser
> 
> ; Password used for connection
> ;password = mypassword
> 
> ; Topic where the meters data as JSON string is published
> ; minimum required JSON payload: { "Dc": { "Power": 321.6, "Voltage": 52.6  }, "Soc": 63 }
> topic = enphase/battery
> ```
> 
>  ```json
> {
>     "Dc": {
>         "Power": 321.6,                       
>         "Voltage": 52.7,                      
>         "Current": 6.10,                      
>         "Temperature": 23                     
>     },
>     "InstalledCapacity": 200.0,               
>     "ConsumedAmphours": 74.5,                 
[... 87 lines cut ...]
>         "Cell11": 0,                          
>         "Cell12": 0,                         
>         "Cell13": 0,                          
>         "Cell14": 0,                         
>         "Cell15": 0,                          
>         "Cell16": 0,                          
>         "Cell17": 0,                          
>         "Cell18": 0,                          
>         "Cell19": 0,                          
>         "Cell20": 0,                          
>         "Cell21": 0,                          
>         "Cell22": 0,                          
>         "Cell23": 0,                          
>         "Cell24": 0                           
>     },
>     "Io": {
>         "AllowToCharge": 0,                   
>         "AllowToDischarge": 0,                
>         "AllowToBalance": 0,                  
>         "ExternalRelay": 0                    
>     }
> }
> ```

Wait… is this all one file? If yes you have to remove the JSON. This is only an example you have to send via MQTT.

harald7, 2023-04-05:
Good morning Manuel

Yes, it was all in the config.ini

After removing the json it's working.

`2023-04-05 07:44:40.042999500 *** starting dbus-mqtt-battery ***
2023-04-05 07:44:41.234346500 INFO:root:MQTT client: Connected to MQTT broker!
2023-04-05 07:44:41.238671500 INFO:root:Waiting 5 seconds for receiving first data...
2023-04-05 07:44:46.244143500 INFO:root:Waiting 5 seconds for receiving first data...
2023-04-05 07:44:51.248650500 INFO:root:Waiting 5 seconds for receiving first data...
2023-04-05 07:44:56.253985500 INFO:root:Waiting 5 seconds for receiving first data...
2023-04-05 07:45:01.259554500 INFO:root:Waiting 5 seconds for receiving first data...
2023-04-05 07:45:06.263210500 INFO:root:Waiting 5 seconds for receiving first data...
2023-04-05 07:45:11.268788500 INFO:root:Waiting 5 seconds for receiving first data...
2023-04-05 07:45:33.707125500 INFO:root:Waiting 5 seconds for receiving first data...
2023-04-05 07:45:38.712683500 INFO:root:Waiting 5 seconds for receiving first data...
2023-04-05 07:45:43.718254500 INFO:root:Waiting 5 seconds for receiving first data...
2023-04-05 07:45:48.723811500 INFO:root:Waiting 5 seconds for receiving first data...
2023-04-05 07:45:53.729375500 INFO:root:Waiting 5 seconds for receiving first data...
2023-04-05 07:45:58.735009500 WARNING:root:Waiting since 60 seconds for receiving first data...
2023-04-05 07:46:03.740567500 INFO:root:Waiting 5 seconds for receiving first data...
2023-04-05 07:46:08.746130500 INFO:root:Waiting 5 seconds for receiving first data...
2023-04-05 07:46:13.751693500 INFO:root:Waiting 5 seconds for receiving first data...
2023-04-05 07:46:18.757253500 INFO:root:Waiting 5 seconds for receiving first data...
2023-04-05 07:46:23.762821500 INFO:root:Waiting 5 seconds for receiving first data...
`
Thanks a lot!

Another question if it's OK to ask in this issue.
I have all the data in ioBroker, the NodeRed adapter is also installed.
What's the easiest way to create the json string?

mr-manuel, 2023-04-05:
Unfortunately I don't know ioBroker, but it should also be possible to generate the JSON directly there. I would use Node-RED, if you don't know both.

harald7, 2023-04-05:
Do you have a link for a example in Node-Red?

mr-manuel, 2023-04-05:
Unfortunately not, but if you search for "node red send json over mqtt" on Google you find a lot.
Probably you have also to join the data you get from ioBroker.

closed 2023-04-05 by mr-manuel