mr-manuel/venus-os_dbus-mqtt-battery #11: Charge Current Limit based on Time of Day + General Battery Monitor Question

J4nsen, 2023-04-14:
Hi,
I found this repo through your link in a dbus-serial-battery issue and was wondering, if I could use them together.

Basically I want to limit the Charge/Discharge Current based on the time of day. My problem is the noise my three Multiplus 2s create when turning on the fans, and since my bedroom is located next to them, preventing the fans to turn on during my sleep would be nice. 

However, (atm) I only want to change the Charge and Discharge Current through Node-Red and let the other logic reside in dbus-serial-battery. Is this possible? 

This also leads to my next question: Currently I have the Multiplus as my Battery Monitor selected. However dbus-serial-battery still imposes its parameters. Who is actually in charge of the limits?

Thanks in advance, Jan

J4nsen, 2023-04-14:
Hey,
thanks for your fast answer and thanks for your suggestion. I will test the Victron Energy Nodes. :)

Yes, I've enabled ESS and DVCC. But I dont understand why dbus-serialbattery works, when i've set the battery monitor to the Multiplus. 

Anyway, this issue has nothing to do with your project.

mr-manuel, 2023-04-14:
Then this is the node you are searching for:

[image]

Why should dbus-serialbattery not work? You can have more than one battery monitore. Some have even more than 5 batteries.

J4nsen, 2023-04-14:
Thanks!

You said

>> if you have selected the MultiPlus as battery monitor, than the MultiPlus set the CCL and DCL.

and I assumed that with the multiplus as the monitor, dbus-serialbatterys settings regarding CCL and DCL would have no effect. How would different monitors work if they calculate/set different values for CCL and DCL?

mr-manuel, 2023-04-14:
It gets only the value from the selected monitor.

pos-ei-don, 2023-04-14:
> It gets only the value from the selected monitor.

I think that this is only mostly true.
When a single BMS lowers the Max Charge Current,  I belive the lowest one is used. I bet I saw this behabiour several Times...

mr-manuel, 2023-04-14:
Maybe best would be, if you ask in the victron community.

J4nsen, 2023-04-15:
Thanks 👍

I will also post my findings there. Probably tomorrow

mr-manuel, 2023-04-15:
Ok, so the Victron Community got an answer :)
The battery monitor is not the same as the BMS selected.

With Venus OS >3.00 you can select the BMS that should be used for DVCC:
[image]

J4nsen, 2023-04-15:
Thank you so much for the research! So everything works as expected by the Victron engineers :D

I'm still unsure how I will implement my idea. I will have a look at the ESS Control Node :)

closed 2023-04-15 by J4nsen

J4nsen, 2023-04-29:
The ESS Control Node works nicely. Thanks again :)

[image]

mr-manuel, 2023-04-29:
Nice, can you explain more detailed what exactly you did? Maybe this could also be interesting for others :-)

J4nsen, 2023-04-29:
Sure :)

I started with updating my Venus Controller to the large image to get NodeRed.

In NodeRed I created my flow to set the maximum charge current to 50A based on the day of time. It's simple:
[image]

The first node is [light scheduler](https://flows.nodered.org/node/node-red-contrib-light-scheduler): 
[image]

It outputs ON/OFF as the payload, which in the end sends either to 50, or -1 to the "ESS Control" Node. The value -1 disables the current limit.
[image]

Here my exported flow: [ESS-Current-by-daytime.txt](https://github.com/mr-manuel/venus-os_dbus-mqtt-battery/files/11360022/ESS-Current-by-daytime.txt)