mr-manuel/venus-os_dbus-mqtt-battery #14: BIG Success, but few things to iron out issue #2

seamaster101, 2023-05-09:
Issue #2 
When drivers is working we get the cells that are populated in the MQTT, but all of the fields that should be calculated automatically are not:
[image]
I must be missing something... Hint please?

mr-manuel, 2023-05-09:
Please post a sampe of a MQTT message you are publishing.

seamaster101, 2023-05-09:
@mr-manuel 
here is the housebatterybank MQTT:
`{"Dc":{"Power":204.3336,"Voltage":13.443,"Current":15.2,"Temperature":20},"Soc":100,"InstalledCapacity":900,"Alarms":{"HighVoltage":0,"LowVoltage":0,"HighTemperature":0,"LowTemperature":0},"Info":{"MaxChargeVoltage":14.4,"MaxDischargeCurrent":300,"MaxChargeCurrent":200}}`

here is the battery-1 MQTT:
`{"Dc":{"Power":67.21000000000001,"Voltage":13.442,"Current":5,"Temperature":19},"Soc":100,"InstalledCapacity":300,"Voltages":{"Cell1":3.36,"Cell2":3.36,"Cell3":3.36,"Cell4":3.362},"Alarms":{"LowCellVoltage":0,"HighVoltage":0,"LowVoltage":0,"HighTemperature":0,"LowTemperature":0},"Info":{"MaxChargeVoltage":14.4,"MaxDischargeCurrent":120,"MaxChargeCurrent":100}}`

here is battery-2 mqtt:
`{"Dc":{"Power":238.64875,"Voltage":13.445,"Current":17.75,"Temperature":20},"Soc":100,"InstalledCapacity":900,"Alarms":{"HighVoltage":0,"LowVoltage":0,"HighTemperature":0,"LowTemperature":0},"Info":{"MaxChargeVoltage":14.4,"MaxDischargeCurrent":300,"MaxChargeCurrent":200}}`

here is battery-3 MQTT:
`{"Dc":{"Power":67.21000000000001,"Voltage":13.442,"Current":5,"Temperature":19},"Soc":100,"InstalledCapacity":300,"Voltages":{"Cell1":3.36,"Cell2":3.36,"Cell3":3.36,"Cell4":3.362},"Alarms":{"LowCellVoltage":0,"HighVoltage":0,"LowVoltage":0,"HighTemperature":0,"LowTemperature":0},"Info":{"MaxChargeVoltage":14.4,"MaxDischargeCurrent":120,"MaxChargeCurrent":100}}`

closed 2023-05-10 by mr-manuel

seamaster101, 2023-05-10:
@mr-manuel Thanks for the quick response!
What is the easiest way to update the 4 drivers i have?

mr-manuel, 2023-05-10:
Update one and copy over the `dbus-mqtt-battery.py` to the others. Or just overwrite them while downloading:

```bash
wget -O /data/etc/dbus-mqtt-battery/dbus-mqtt-battery.py https://raw.githubusercontent.com/mr-manuel/venus-os_dbus-mqtt-battery/master/dbus-mqtt-battery/dbus-mqtt-battery.py
wget -O /data/etc/dbus-mqtt-battery-1/dbus-mqtt-battery-1.py https://raw.githubusercontent.com/mr-manuel/venus-os_dbus-mqtt-battery/master/dbus-mqtt-battery/dbus-mqtt-battery.py
wget -O /data/etc/dbus-mqtt-battery-2/dbus-mqtt-battery-2.py https://raw.githubusercontent.com/mr-manuel/venus-os_dbus-mqtt-battery/master/dbus-mqtt-battery/dbus-mqtt-battery.py
wget -O /data/etc/dbus-mqtt-battery-3/dbus-mqtt-battery-3.py https://raw.githubusercontent.com/mr-manuel/venus-os_dbus-mqtt-battery/master/dbus-mqtt-battery/dbus-mqtt-battery.py
```

Then reboot.

seamaster101, 2023-05-10:
Success for the most part! there are still some blank cells. Time-to-go and  Min/Max cell temperature
[image]
[image]

mr-manuel, 2023-05-10:
You need to provide either
`InstalledCapacity` and `ConsumedAmphours`
or
`InstalledCapacity` and `Capacity`

```
    "InstalledCapacity": 200.0,               --> Ampere hours - total battery capacity
    "ConsumedAmphours": 74.5,                 --> Ampere hours - consumed (only positive values)
    "Capacity": 125.5,                        --> Ampere hours - remaining (only positive values) - if missing in the JSON, than gets calculated when "InstalledCapacity" and "ConsumedAmphours" are set
    "TimeToGo": 43967,                        --> Seconds - time until the battery is empty - if missing in the JSON, than gets calculated when "Capacity" is set or calculated
```

I could add, that is calculated from SoC, if not available.

seamaster101, 2023-05-10:
@mr-manuel I'm not quite clear on the logic on the calculations. Just for test I entered values of `ConsumedAmphours ` on `Battery 1 `and `Capacity` on another to see the results. It was surprising to me and in my opinion wrong:
for `Battery 1 ` I entered:
`    Soc: msg.payload.col78,
    InstalledCapacity : 300,
    Capacity : 300,`
SOC is coming from the BMS. The charging current/consumption is about 4.5A (charging) and time to go shows as 0s[image]
for my battery bank I entered:
`    Soc: Math.abs((msg.payload.col40+msg.payload.col40+msg.payload.col40)/3),
    InstalledCapacity : Math.abs(300+300+300),
    ConsumedAmphours : 20,`
In my opinion that leaves me with 880AHrs. With current consumption  about 15 A (charging) it was showing me 1.5hrs (maybe I guess this is the time to fully charge?) 
Then the system went to a float mode and the current become negative -2.8 and is showing me TTG 9D19H. 
This appear to be approx correct, but I don't understand what will happen when the system is rebooted if let's say the consigned amp hours are realistically changed - maybe system restarted when the actual consumed amp hours were 500. Then on restart it will read the hardcoded value of 20AHr consumed and all that calculation will be not valid any more...
also though the discharge process as the consumed AHrs increase, the reading will be still 20.0Ahrs
Please explain how it should operate...I dont get it!
[image]

mr-manuel, 2023-05-10:
When you are discharging it shows you the time until the % value you specified in the config file. If you are charging it shows you the time it takes to reach 100%.

seamaster101, 2023-05-10:
I still can not understand : 
 I don't understand what will happen when the system is rebooted if let's say the consigned amp hours are realistically changed - maybe system restarted when the actual consumed amp hours were 500. Then on restart it will read the hardcoded value of 20AHr consumed and all that calculation will be not valid any more...
also though the discharge process as the consumed AHrs increase, the reading will be still 20.0Ahrs
Please explain how it should operate...I dont get it!

mr-manuel, 2023-05-10:
Sorry but I don‘t get the problem. Your examples are not very clear, only half filled with values and therefore hard to understand. Please describe situation 1, the corresponding JSON with filled values. Then describe situation 2, the corresponding JSON with filled values and so on.

It doesn‘t matter how the values change, they are calculated on every received message. Only the Time to Go is calculated every x seconds (based on the value you set in the config file). This prevents that the value is jumping around,

seamaster101, 2023-05-10:
@mr-manuel Here is very simple example:
1. the BMS reports 100% charged battery 
2. the Battery is charging (current is positive)
3. TTG shows 1d 7h 
I presumed this is the time to charge to 100% but it is already reported by the BMS at 100%
what number your calculations are trying to hit?
[image]

here is the json:
`{"Dc":{"Power":19.399549999999998,"Voltage":13.379,"Current":1.45,"Temperature":20},"Soc":100,"InstalledCapacity":900,"Capacity":800,"Alarms":{"HighVoltage":0,"LowVoltage":0,"HighTemperature":0,"LowTemperature":0},"Info":{"MaxChargeVoltage":14.4,"MaxDischargeCurrent":300,"MaxChargeCurrent":200}}`

mr-manuel, 2023-05-10:
There are `100 Ah` left to charge with `1.45 A`. `"InstalledCapacity":900,"Capacity":800`

https://github.com/mr-manuel/venus-os_dbus-mqtt-battery/blob/f11c777255742ffd602720d6fdaa40a0b9f7e634/dbus-mqtt-battery/dbus-mqtt-battery.py#L369-L417

mr-manuel, 2023-05-10:
In the driver there is no hard coded capacity.

seamaster101, 2023-05-10:
It was hardcoded in my json