mr-manuel/venus-os_dbus-mqtt-battery #16: Doesn't start (automatically) - but .py seems to work. Logs are empty?!

WaldemarFech, 2023-05-11:
I've ran the install.sh - but the driver won't start.
When i run 
`root@einstein:/service/dbus-mqtt-battery# svstat /service/dbus-mqtt-battery`
then i get
`/service/dbus-mqtt-battery: up (pid 2413) 1 seconds`

Log is empty, even with INFO or DEBUG set

But when i run
`python /data/etc/dbus-mqtt-battery/dbus-mqtt-battery.py`
then it is starting and running.

Did i made something wrong?

mr-manuel, 2023-05-11:
Could you post the output of this commands?

```bash
cat /data/rc.local
```

```bash
ls -l /service
```

```bash
ls -l /data/etc/dbus-mqtt-battery
```

```bash
ls -l /data/etc/dbus-mqtt-battery/service
```

WaldemarFech, 2023-05-11:
```shell
root@einstein:~# cat /data/rc.local
#!/bin/bash
#ln -s /data/dbus-aggregate-batteries/service /service/dbus-aggregate-batteries
sh /data/etc/dbus-serialbattery/reinstall-local.sh
bash /data/etc/dbus-mqtt-battery/install.sh
```
```shell
root@einstein:~# ls -l /service
drwxr-xr-x    4 root     root           100 May  9 12:04 PackageManager
lrwxrwxrwx    1 root     root            29 May  9 12:05 can-bus-bms.can1 -> /run/service/can-bus-bms.can1
drwxr-xr-x    4 root     root           100 May  9 12:04 dbus-adc
drwxr-xr-x    4 root     root           120 May  9 12:04 dbus-ble-sensors
lrwxrwxrwx    1 root     root            42 May  9 12:05 dbus-cgwacs.ttyUSB0 -> /var/volatile/services/dbus-cgwacs.ttyUSB0
lrwxrwxrwx    1 root     root            42 May  9 12:04 dbus-cgwacs.ttyUSB1 -> /var/volatile/services/dbus-cgwacs.ttyUSB1
lrwxrwxrwx    1 root     root            42 May  9 12:05 dbus-cgwacs.ttyUSB2 -> /var/volatile/services/dbus-cgwacs.ttyUSB2
drwxr-xr-x    4 root     root           100 May  9 12:04 dbus-digitalinputs
drwxr-xr-x    4 root     root           100 May  9 12:04 dbus-fronius
drwxr-xr-x    4 root     root           100 May  9 12:04 dbus-generator-starter
drwxr-xr-x    4 root     root           100 May  9 12:04 dbus-modbus-client
lrwxrwxrwx    1 root     root            22 May  9 12:05 dbus-mqtt -> /run/service/dbus-mqtt
lrwxrwxrwx    1 root     root            35 May  9 13:41 dbus-mqtt-battery -> /data/etc/dbus-mqtt-battery/service
drwxr-xr-x    4 root     root           120 May  9 12:04 dbus-pump
drwxr-xr-x    4 root     root           100 May  9 12:04 dbus-qwacs
lrwxrwxrwx    1 root     root            49 May  9 12:04 dbus-serialbattery.ttyUSB0 -> /var/volatile/services/dbus-serialbattery.ttyUSB0
lrwxrwxrwx    1 root     root            49 May  9 12:05 dbus-serialbattery.ttyUSB2 -> /var/volatile/services/dbus-serialbattery.ttyUSB2
drwxr-xr-x    4 root     root           100 May  9 12:04 dbus-systemcalc-py
drwxr-xr-x    4 root     root           120 May  9 12:04 dbus-tempsensor-relay
drwxr-xr-x    4 root     root           100 May  9 12:04 dbus-vebus-to-pvinverter
lrwxrwxrwx    1 root     root            39 May  9 12:05 gps-dbus.ttyUSB0 -> /var/volatile/services/gps-dbus.ttyUSB0
lrwxrwxrwx    1 root     root            39 May  9 12:05 gps-dbus.ttyUSB2 -> /var/volatile/services/gps-dbus.ttyUSB2
drwxr-xr-x    4 root     root           100 May  9 12:04 gui
lrwxrwxrwx    1 root     root            20 May  9 12:05 hostapd -> /run/service/hostapd
drwxr-xr-x    4 root     root           100 May  9 12:04 hub4control
drwxr-xr-x    4 root     root           100 May  9 12:04 llmnrd
drwxr-xr-x    4 root     root           100 May  9 12:04 localsettings
lrwxrwxrwx    1 root     root            37 May  9 12:04 mk2-dbus.ttyS4 -> /var/volatile/services/mk2-dbus.ttyS4
lrwxrwxrwx    1 root     root            22 May  9 12:05 mosquitto -> /run/service/mosquitto
lrwxrwxrwx    1 root     root            21 May  9 12:05 mqtt-rpc -> /run/service/mqtt-rpc
drwxr-xr-x    4 root     root           100 May  9 12:04 netmon
drwxr-xr-x    4 root     root           100 May  9 12:04 nginx
drwxr-xr-x    4 root     root           120 May  9 12:04 node-red-venus
lrwxrwxrwx    1 root     root            20 May  9 12:04 openssh -> /run/service/openssh
drwxr-xr-x    4 root     root           120 May  9 12:04 ppp
drwxr-xr-x    4 root     root           100 May  9 12:04 serial-starter
drwxr-xr-x    4 root     root           100 May  9 12:04 service-advertiser
drwxr-xr-x    4 root     root           120 May  9 12:04 signalk-server
drwxr-xr-x    4 root     root           100 May  9 12:04 simple-upnpd
lrwxrwxrwx    1 root     root            23 May  9 12:04 ssh-tunnel -> /run/service/ssh-tunnel
lrwxrwxrwx    1 root     root            28 May  9 12:04 vecan-dbus.can0 -> /run/service/vecan-dbus.can0
lrwxrwxrwx    1 root     root            47 May  9 12:04 vedirect-interface.ttyS5 -> /var/volatile/services/vedirect-interface.ttyS5
lrwxrwxrwx    1 root     root            47 May  9 12:04 vedirect-interface.ttyS6 -> /var/volatile/services/vedirect-interface.ttyS6
lrwxrwxrwx    1 root     root            47 May  9 12:04 vedirect-interface.ttyS7 -> /var/volatile/services/vedirect-interface.ttyS7
lrwxrwxrwx    1 root     root            49 May  9 12:05 vedirect-interface.ttyUSB0 -> /var/volatile/services/vedirect-interface.ttyUSB0
lrwxrwxrwx    1 root     root            49 May  9 12:05 vedirect-interface.ttyUSB2 -> /var/volatile/services/vedirect-interface.ttyUSB2
drwxr-xr-x    4 root     root           100 May  9 12:04 venus-access
drwxr-xr-x    4 root     root           100 May  9 12:04 venus-button-handler
drwxr-xr-x    4 root     root           100 May  9 12:04 venus-html5-logger
drwxr-xr-x    4 root     root           100 May  9 12:04 venus-platform
lrwxrwxrwx    1 root     root            27 May  9 12:05 vesmart-server -> /run/service/vesmart-server
drwxr-xr-x    4 root     root           100 May  9 12:04 vrmlogger
lrwxrwxrwx    1 root     root            25 May  9 12:05 websockify-c -> /run/service/websockify-c
```

```shell
root@einstein:~# ls -l /data/etc/dbus-mqtt-battery
-rw-r--r--    1 root     root         13466 May 11 09:46 battery_service.py
-rw-r--r--    1 root     root          1848 May 11 08:45 config.ini
-rw-r--r--    1 root     root          1833 May  9 13:03 config.sample.ini
-rwxr--r--    1 root     root         27455 May 11 07:10 dbus-mqtt-battery.py
[... cut ...]
-rwxr--r--    1 root     root           264 May  9 13:40 uninstall.sh
```
(Don't wonder about the size of battery_service.py. I have tested a bit after it haven't started)

```shell
root@einstein:~# ls -l /data/etc/dbus-mqtt-battery/service
drwxr-xr-x    3 root     root          4096 May  9 13:12 log
-rwxr-xr-x    1 root     root           120 May  9 13:03 run
lrwxrwxrwx    1 root     root            35 May  9 13:24 service -> /data/etc/dbus-mqtt-battery/service
drwx------    2 root     root          4096 May 11 16:31 supervise
```

mr-manuel, 2023-05-11:
Somehow there is a loop in your folders. Please run the uninstall script, delete the /data/etc/dbus-mqtt-battery folder and download it again

closed 2023-09-12 by mr-manuel