mr-manuel/venus-os_dbus-mqtt-battery #17: MQTT Battery and Aggregate Battery possible?

Zahnpasta83, 2023-09-23:
Is there a way to integrate data from a battery based on MQTT Battery together with batteries based on Serial Battery in Aggregate Battery?

pos-ei-don, 2023-09-23:
Of course! Take a look at Manuels repository.

Zahnpasta83, 2023-09-23:
🤷🏻‍♂️ which one? Can you link it?

pos-ei-don, 2023-09-23:
What is wrong with this one you are currently in?

mr-manuel, 2023-09-23:
If you aleady have installed [`dbus-mqtt-battery`](https://github.com/mr-manuel/venus-os_dbus-mqtt-battery) and [`dbus-serialbattery`](https://github.com/Louisvdw/dbus-serialbattery) they should be both visible in [dbus-aggregate-batteries](https://github.com/Dr-Gigavolt/dbus-aggregate-batteries). If not you could try [BatteryAggregator](https://github.com/pulquero/BatteryAggregator).

mr-manuel, 2023-09-25:
Linked issue: https://github.com/Dr-Gigavolt/dbus-aggregate-batteries/issues/51

closed 2023-09-25 by mr-manuel