mr-manuel/venus-os_dbus-mqtt-battery #26: Device not present

dpritt, 2024-08-15:
Running Venus OS v3.50~13 (this is a beta of the new Gui, but I am viewing the old console) on a Raspberry Pi 3.
This is a test build, so nothing else connected at the moment.
Large image - Node Red enabled.

Installed the device driver, everything seemed correct, changed the config.ini to use the IP address of the Raspberry Pi.
The MQTT driver is not showing in the 'Devices' section.
Opening Node Red I cannot find a battery in the relevant node.
Log shows    *** starting dbus-mqtt-battery ***
                     WARNING: root:Waiting since 60 seconds for receiving first data....

But as I cannot locate the battery monitor in Node Red I cannot create any input.

Any suggestions?

mr-manuel, 2024-08-15:
Probably you understood wrong how the driver works. The driver fetches the data from the MQTT broker and topic you defined in the `config.ini`. You can use Node-RED to post the correct JSON data to the topic.

dpritt, 2024-08-15:
Manuel,
 
Yes, I was expecting to see the Virtual Driver in Node-Red as I was going to manipulate the MQTT payload before sending it through.
 
What I was hoping to do is show the voltages of the two starter batteries and the generator battery on my Cerbo GX, currently it only monitors my domestic battery.
 
The other battery voltages are published using MQTT, and I was going to send dummy data for the SOC and Current values as I don't calculate them, adding this in Node-Red, that is why I was looking for the device in the Node-Red objects.
 
I will create some data to see if that will make the device appear, and if so republish the data via MQTT to your device driver.
 
I was using an alternative I found, but this only supports a single virtual battery, and I actually need three (or 4 if I include the bow-thruster batteries).
 
Many thanks for explaining how it works.
 
David

closed 2024-08-19 by mr-manuel