mr-manuel/venus-os_dbus-mqtt-battery #27: Logic advise for discharging battery

kopierschnitte, 2024-09-14:
Hi!

Thanks for the great project. It enabled me to make my non-Victron ESS visible within VRM.

At the moment, I don't really understand how charging/discharging is handled by your driver or, better said, by VenusOS.

If my external battery is charged, VenusOS automatically adds this to the AC-Load of my entire installation and if it's being discharged, I needed to "fake" solar production with a custom inverter driver in order to have correct AC-Loads.

This of course disturbs my solar statistics and therefore also the production forecast.

Do you see any possibility that your driver could also correctly "inject" charging/discharging? Or didn't I understand the logic correctly at this point?

mr-manuel, 2024-09-14:
You need a MultiPlus to achieve that, since you want to merge DC with AC.

See https://github.com/mr-manuel/venus-os_dbus-multiplus-emulator

kopierschnitte, 2024-09-14:
Well, I do have a MP2 with a connected 48V battery ;-)

mr-manuel, 2024-09-14:
Then you probably have a configuration issue somewhere. In this case ask your question in the [Victron Community](https://community.victronenergy.com/).

closed 2024-09-14 by mr-manuel

kopierschnitte, 2024-09-14:
Thanks! But do you think, VenusOS even respects those kind of (dbus-connected) batteries or does it just displays the statistics without doing anything with the data?

mr-manuel, 2024-09-14:
It doesn't matter how you connect a battery in Venus OS. When it is recognized, all batteries should display the same way.

kopierschnitte, 2024-09-14:
Okay, then I suspect that something's wrong if my battery is shown like this in VRM, isn't it?
[image]
The Cerbo itself displays it as expected:
[image]

mr-manuel, 2024-09-14:
See https://github.com/mr-manuel/venus-os_dbus-mqtt-battery/issues/27#issuecomment-2350935676