mr-manuel/venus-os_dbus-mqtt-battery #30: driver does start with error after reboot and exit

teefixx, 2024-10-06:
Hi al

See my log. After reboot the driver exits with an error and I need to restart it manually. After restart everything works. 
Could this behaviour be caused by long booting time and node red coming up late? I write all the values for the driver to the MQTT broker via node red...

2024-10-06 15:10:31.643692500 INFO:root:Waiting 5 seconds for receiving first data...
2024-10-06 15:10:36.649220500 INFO:root:Waiting 5 seconds for receiving first data...
2024-10-06 15:10:41.653626500 WARNING:root:Waiting since 240 seconds for receiving first data...
2024-10-06 15:10:46.656037500 INFO:root:Waiting 5 seconds for receiving first data...
2024-10-06 15:10:51.661666500 INFO:root:Waiting 5 seconds for receiving first data...
2024-10-06 15:10:56.667268500 INFO:root:Waiting 5 seconds for receiving first data...
2024-10-06 15:10:57.612742500 WARNING:root:Received JSON doesn't contain minimum required values
2024-10-06 15:10:57.613186500 WARNING:root:Example: {"Dc":{"Power":321.6,"Voltage":52.7},"Soc":63}
2024-10-06 15:11:01.671435500 INFO:root:Waiting 5 seconds for receiving first data...
2024-10-06 15:11:05.745824500 WARNING:root:Received key "/Dc/0/Power" with value "None" is not valid
2024-10-06 15:11:05.746175500 WARNING:root:Received key "/Dc/0/Voltage" with value "None" is not valid
2024-10-06 15:11:05.746519500 WARNING:root:Received key "/Dc/0/Current" with value "None" is not valid
2024-10-06 15:11:05.746850500 WARNING:root:Received key "/InstalledCapacity" with value "None" is not valid
2024-10-06 15:11:05.747184500 WARNING:root:Received key "/ConsumedAmphours" with value "None" is not valid
2024-10-06 15:11:05.747509500 WARNING:root:Received key "/Capacity" with value "None" is not valid
2024-10-06 15:11:05.747828500 WARNING:root:Received key "/Soc" with value "None" is not valid
2024-10-06 15:11:06.679480500 INFO:root:Waiting 5 seconds for receiving first data...
2024-10-06 15:11:11.685107500 INFO:root:Waiting 5 seconds for receiving first data...
2024-10-06 15:11:14.362430500 WARNING:root:Received key "/Dc/0/Voltage" with value "None" is not valid
2024-10-06 15:11:14.362808500 WARNING:root:Received key "/Dc/0/Current" with value "None" is not valid
2024-10-06 15:11:14.363255500 WARNING:root:Received key "/InstalledCapacity" with value "None" is not valid
2024-10-06 15:11:14.363657500 WARNING:root:Received key "/ConsumedAmphours" with value "None" is not valid
2024-10-06 15:11:14.364030500 WARNING:root:Received key "/Capacity" with value "None" is not valid
2024-10-06 15:11:14.364381500 WARNING:root:Received key "/Soc" with value "None" is not valid
2024-10-06 15:11:16.740855500 INFO:root:registered ourselves on D-Bus as com.victronenergy.battery.mqtt_battery_15
2024-10-06 15:11:16.744152500 INFO:root:Connected to dbus and switching over to GLib.MainLoop() (= event based)
2024-10-06 15:11:17.748391500 Traceback (most recent call last):
2024-10-06 15:11:17.748396500   File "/data/etc/dbus-mqtt-battery/dbus-mqtt-battery.py", line 630, in _update
2024-10-06 15:11:17.748400500     "Battery SoC: {:.2f} V - {:.2f} %".format(
2024-10-06 15:11:17.748403500 TypeError: unsupported format string passed to NoneType.__format__

==> here I do the manual restart with restart.sh

2024-10-06 15:14:27.839277500 *** starting dbus-mqtt-battery ***
2024-10-06 15:14:28.940841500 INFO:root:MQTT client: Connecting to broker localhost on port 1883
2024-10-06 15:14:28.970327500 INFO:root:MQTT client: Connected to MQTT broker!
2024-10-06 15:14:28.971351500 INFO:root:Waiting 5 seconds for receiving first data...
2024-10-06 15:14:34.022261500 INFO:root:registered ourselves on D-Bus as com.victronenergy.battery.mqtt_battery_15
2024-10-06 15:14:34.023896500 INFO:root:Connected to dbus and switching over to GLib.MainLoop() (= event based)
2024-10-06 15:14:35.025793500 INFO:root:Battery SoC: 0.00 V - 98.85 %
2024-10-06 15:14:44.041273500 INFO:root:Battery SoC: 0.00 V - 98.85 %
2024-10-06 15:14:55.058360500 INFO:root:Battery SoC: -1.44 V - 98.85 %

Any help would be appreciated.

Thank you

mr-manuel, 2024-10-06:
Yes. Add a check, that Node-RED sends the data only, if the minimum data is present and that values that are still `undefined` or `None` are not send. Check also the messages that are send with MQTT Explorer, so you can better understand what is going on.

teefixx, 2024-10-07:
Hi mr-manuel

Thank you for the advice. That worked, no errors anymore and driver starts up as intended.

closed 2024-10-07 by teefixx

mr-manuel, 2024-10-07:
You're welcome.